linuxmint/mintdrivers #16: b43 not listed as a driver option for BCM4311

JamesTheAwesomeDude, 2015-05-16:
On the Dell Inspiron 1720 laptop, which has a BCM4311 ("Broadcom AirForce 54g 802.11a/b/g PCI Express Transceiver") Wi-Fi chip, the driver manager only lists 2 options: the proprietary STA driver, or nothing.
[image]

However, the Wi-Fi (and, miraculously, the **bluetooth**) work perfectly after simply installing the package `firmware-b43-installer` and rebooting the computer.

But the Driver Manger remains convinced that the device is nonfunctional without the STA driver. (The above screenshot was actually taken while connected to a Wi-Fi network via the supposedly "not working" device!)

It would be nice if the open-source b43 driver were added to the database of working drivers. At least on the BCM4311 chipset, the driver both performs well and has no noticeable bugs. The installation went off without a hitch, as well, which is why I am highly recommending this driver to be included in the database.

This screenshot was taken on 17.1 MATE 64-bit with all updates of level 3 and lower installed as of 5/16/2015. The system had been installed the previous day.

clefebvre, 2015-06-11:
Hi,

This was fixed and then reverted by Alberto Milone in the upstream version of ubuntu-drivers-common. We're facing a licensing issue here ... it's really messed up.

- The open-source driver b43 requires non-free firmware for certain chipsets.
- The IDs aren't properly set so we can't accurately identifiy all of them
- linux-firmware-nonfree has licensing issue... apparently it can't be on the live medium

So even though the STA driver doesn't work as well, it's one we can recommend and install for everybody and so it's a better solution here.

Now, with that said... we could probably make the UI show that the device is using b43 rather than saying it's not working. That's another issue though and I think we'd need to modify ubuntu-drivers-common slightly here as well to achieve that.

JamesTheAwesomeDude, 2015-06-11:
The b43 driver requires non-free firmware.. but isn't the whole point of the Driver Manager to expedite installation of drivers, particularly non-free ones which cannot be included by default?

The Wi-Fi and Bluetooth were both NOT working out of the box with the current setup. And the STA driver was not working at all. But the b43 driver worked perfectly for both Wi-Fi and Bluetooth and performed well.

I think that since this tool is designed for the installation of non-free drivers, it would make a lot of sense to allow it to install the b43 drivers.

JamesTheAwesomeDude, 2015-06-18:
> * The open-source driver b43 requires non-free firmware for certain chipsets.

The b43 driver may require "non-free firmware *for certain chipsets*", but the **whole STA driver is closed-source**. That's not an improvement, especially when the STA driver performs worse and is less reliable overall.

> * The IDs aren't properly set so we can't accurately identifiy all of them

Even if the PCI IDs are mis-set, the human-readable card names don't seem to be. `b43` compatibility could be easily and reliably checked by running:

    lspci | grep -oP 'BCM\K43(06|09|11|12|18|22|31|224|225)'

This matches all the cards listed as compatible with the driver under https://help.ubuntu.com/community/WifiDocs/Driver/bcm43xx.

> * linux-firmware-nonfree has licensing issue... apparently it can't be on the live medium

Nor would it need to be in order to be in the Driver Manager's database. The Driver Manager is specifically designed to take care of installing drivers which cannot come with the system by default and may have licensing/installation complications.

JamesTheAwesomeDude, 2015-06-29:
@clefebvre I have addressed all of the points you've brought up. What further obstacles are there to reverting this regression?

clefebvre, 2015-11-07:
It's not that simple. We're unable to assess via modalias (the method used in Ubuntu) whether a particular chipset works better with STA or B43, or even if it works at all with B43.

My chipset here for instance does not work at all with B43, but does with STA.
Other chipsets work better with B43.
Other chipsets work better with B43+linux-firmware-nonfree but not at all with B43 alone.

Alberto's plan wasn't just to put linux-firmware-nonfree on the live media, it was to add all BCM firmware modules in linux-firmware-nonfree.. and that was rejected.

Now with all that said, and although it won't work for people who aren't connected to the Internet, we'll modify ubuntu-drivers-common in 17.3 to not only list bcwml-kernel-source but also firmware-b43-installer and firmware-b43legacy-installer.

clefebvre, 2015-11-07:
https://github.com/linuxmint/ubuntu-drivers-common/commit/864151c6b0c4a75cdf9d2f0ebc968f42e508082f

closed 2015-11-07 by clefebvre

JamesTheAwesomeDude, 2015-11-15:
>with all that said, and although it won't work for people who aren't connected to the Internet, we'll modify ubuntu-drivers-common in 17.3 to not only list bcwml-kernel-source but also firmware-b43-installer and firmware-b43legacy-installer.

Thanks! :relieved: :+1:  
That's one more thing off my mind when recommending Mint to people.