linuxmint/mintdrivers #21: [Feature Request] Driver Manager showing newest Proprietary GPU Drivers

Mintberry1, 2016-01-20:
JosephMcc from the cinnamon section:
"This isn't a Cinnamon issue. Report it here https://github.com/linuxmint/mintdrivers"

The Driver Manager doesnt show "Driver Release Dates". If the Release Dates would be added you could at least discern how recent the driver is you have. (As a Mint User Noob i even thought that the driver is 2 years old since Ubuntu 14.04 release, adding a release date would have helped immensely)

Please make it so that the "Driver Manager" ALLOWS you to install the LATEST AMD GPU drivers. The only Option as of now is 2:15.200-0ubuntu0.5 (current ~3.315) which seems to be at least 6 months old. The newest driver on AMD is release date 12/17/2015 and isnt even shown in the Driver Manager at all. (At least you could put it there as an option marked [EXPERIMENTAL] if stability concerns is, what is preventing you until now, to not add them to the "Driver Manager".

This is the most important Feature for making Linux Mint more gamer and noob friendly. Now that Steam (Valve) supports Linux with their SteamOS, Linux becomes interesting for converting Windows Users but the most difficult and lacking feature of Linux Mint is that you cant install the latest drivers easily. If you want to install the drivers from AMD's site you have not only go to the CONSOLE and start it manually, you have to uninstall by console old drivers as well. Another thing is that because Linux Mint is based on Ubuntu i assumed that i could install the latest Ubuntu 14.04 Driver. But i was wrong and it broke my Linux Mint. Now i use the generic Linux driver and it works. But as you can see it is very complicated and there is so much things which can wrong manually installing latest drivers especially because there were no tutorials for AMD installation and any good youtube videos. If you care about the future of Linux and especially now that Valve is pushing it a bit more to mainstream you should implement it, that the Driver Manager gives the option of experimental, probably even unstable, latest Linux Drivers on release. Perhaps you can even work with AMD and contact them if necessary so that they help you to make the installation of their drivers more easy to integrate for you as well. And please make it so that the DM shows the Release Date of the driver as well, not only the version.

Adding a Help Button or Tooltips to the Driver Manager (so that you get immediate help explaining things would help as well). Especially what the difference between fglrx and fglrx-updates is for example. Even after googling i still am not sure about the differences. fglrx would suggest to be a fixed driver and the fglrx-updates will update my driver automatically if a new one comes out ?!? Because for now and some months until now they have the same version and it didnt change. (I know i am new but it would help immensely to get a short tooltip explanation on those things at the spot.)  

I dont like Windows but since Windows 95, 20 years ago you could install the newest gpu drivers easily (GUI only). And even when something broke you could go to a safemode and repair it by uninstalling the latest driver. Windows 10 even makes it more easy by installing and finding newest drivers for the Enduser. You must consider the "Dumbest Possible User" and i think in the year 2016 to expect someone to use the console for installing such a trivial thing like the latest GPU driver is very sad.

Mintberry1, 2016-01-20:
Especially in the future with the new VULKAN Graphic API/driver an uptodate driver is a must to compete with SteamOS. SteamOS will certainly add automatic latest gpu drivers support and so should Mint as well. 6 months old drivers (for now could be some 3 months or more until new ones are added so 9+ months old drivers in total doesnt cut it. 

Graphic Card Manufacturers would certainly support Linux more if distributions like Mint allow easily to install latest Experimental GPU drivers. More would use the feature and Nvidia and AMD would get more feedback as well, and would grant even more attention to their drivers when more users use them. Be certain that the download count of proprietary Linux Drivers are monitored. So if somehow statistics would show in Steam and such that Linux Users are mostly using the latest GPU drivers this would cause the manufacturers to pay more attention to their Linux drivers. But if all of us are using 6+ months old drivers then you cant blaim the manufacturers to not invest much into their drivers.

rkingsbury, 2016-02-07:
It would be really nice if Mint would automatically enable the [NVIDIA driver PPA](https://launchpad.net/~graphics-drivers/+archive/ubuntu/ppa) and display the available options.

fizzgigfuzzlehead, 2016-05-03:
I totally agree with that. Would be great to have current drivers on my system without the need to add a PPA to the system.

Maybe it could be like the new update manager for Mint 18. You could choose if you would like to use stable drivers or automatically update to the newest one.

closed 2022-09-12 by clefebvre