linuxmint/mintdrivers #27: Non-root execution via gksu requires -k to pass required environment

javabrett, 2016-06-02:
$ dpkg-query -W mintdrivers
    mintdrivers	1.1.6

Something may have changed in recent times to require environment to be passed to Python `Gtk` main, and such environment is not passed by-default when running as non-root via the script-initiated `gksu`, because it does not pass the `-k` flag.  The result is that the above version now exits silently status `133`.

Adding -k will fix this.  As a workaround, you can run `gksu` manually:

    $ gksu -k /usr/lib/linuxmint/mintDrivers/mintDrivers.py

clefebvre, 2016-06-16:
Hi,

I can't reproduce this issue. What environment do you test this with?

Also, can you describe the issue from a user point of view? You mean the driver manager doesn't actually start? Or if it does, what is the impact of the missing environment?

javabrett, 2016-06-16:
System:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	LinuxMint
Description:	Linux Mint 17.1 Rebecca
Release:	17.1
Codename:	rebecca

$ dpkg-query -W mintdrivers gksu
gksu	2.0.2-6ubuntu2
mintdrivers	1.1.6
```

Symptoms are that having selected _Driver Manager_ from the menu, you get the `gksu` prompt, enter a valid password, then nothing.  If you instead run `mintdrivers` from a terminal, same thing, the window never appears.  If you instead run:

    $ /usr/lib/linuxmint/mintDrivers/mintDrivers.py
    $ echo $?
    133
 
Running in a root shell gives a hint on what environment it might be missing:

    # /usr/lib/linuxmint/mintDrivers/mintDrivers.py
    
    (mintDrivers.py:690): Gdk-ERROR **: error: XDG_RUNTIME_DIR not set in the environment.

    Trace/breakpoint trap

Finally, if you run `xhost+` as the current user, then `sudo su -` for a root shell, to get a good launch you need to run (now as root) e.g.:

    DISPLAY=:0 XDG_RUNTIME_DIR=/run/user/1000 /usr/lib/linuxmint/mintDrivers/mintDrivers.py

This might relate to my `/etc/sudoers`, although I thought it was more permissive on environment than the default, retaining proxies:

    Defaults	env_keep = "http_proxy https_proxy"

Adding `DISPLAY` and `XDG_RUNTIME_DIR` there is also a workaround.

If you are unable to reproduce this problem for 1.1.6, then I guess it must be that somehow these environment variables are being passed through `gksu`, perhaps related to a permissive `env_keep`?

clefebvre, 2016-06-23:
Hmm.. I think it might relate to mdm.. let me get back to you on this.

javabrett, 2016-06-24:
I haven't mentioned all of the significant parts of my `sudoers`:

```
Defaults        env_keep = "http_proxy https_proxy"
Defaults        env_reset
Defaults        mail_badpass
Defaults        secure_path="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"
```

But it doesn't seem to make any difference if `env_reset` is removed ... something seems convinced it needs to scrub `DISPLAY` and `XDG_RUNTIME_DIR`.

Should we be running `gksudo` instead of `gksu`?  They symlink to the same binary, but have arg0 logic as well as reading from `gconf`.  I also read that they might be deprecated in favour of `pkexec`.

javabrett, 2016-06-24:
As an alternative to adding `-k`, we could simply pass the known-required environment on the command-line.

javabrett, 2016-07-17:
Exact cause still unknown, but this seems to be a fairly broad problem, and has started popping-up for me with other commands.  As you suggest, likely related to X or window-manager.

E.g. https://forums.linuxmint.com/viewtopic.php?t=171159

xenopeek, 2018-02-22:
While mintdrivers still uses gksu this issue doesn't appear to have been caused by the use of that. Indeed, running `/usr/lib/linuxmint/mintdrivers/mintdrivers.py` from a root shell does not give the ` XDG_RUNTIME_DIR not set in the environment` it did above. The code has also seen some changes since this issue was reported.

closed 2018-02-22 by xenopeek