linuxmint/mintdrivers #32: bug in sarah wireless card driver

rpietro, 2016-06-14:
not sure whether this is the right place to report a sarah bug, but here we go:

using a dell xps 13 with a broadcom corp bcm4352 802.11ac wireless network adapter, which requires a driver (bcmwl-kernel-source, version 6.30.223.248+bdcom-0ubuntu8 broadcom 802.11 linux sta wireless driver source) from the get go. this worked fine in rosa, could just get it from the USB. but with sarah it will install just fine the first time, but once the computer is rebooted the wifi icon goes back to its pre-driver state and i can no longer see any networks (although the network i chose before is still reachable). to get it back, i have to go back to the driver manager, uninstall and reinstall, reboot and then the wifi sign is back

i can provide additional info if you could guide me on what to send

many thanks

clefebvre, 2016-06-16:
Thanks, this looks like a hardware issue though, so it's not related to the manager itself.

closed 2016-06-16 by clefebvre

rpietro, 2016-06-16:
Clement, thanks. It was working semalessly in Rosa, so not sure this is a hardware issue. Is there anything else I might provide to try to identify where the problem is coming from? It might also be a problem with the applet, I guess. 

thanks a ton