linuxmint/mintdrivers #36: Wrong info in the driver manager

Cos-mo, 2016-11-07:
Here is the inxi-report regarding my Graphic:
`Graphics:  Card-1: Intel 4th Gen Core Processor Integrated Graphics Controller
           Card-2: NVIDIA GK106M [GeForce GTX 760M]
           Display Server: X.Org 1.15.1 drivers: nvidia,intel
           Resolution: 1920x1080@60.0hz
           GLX Renderer: GeForce GTX 760M/PCIe/SSE2
           GLX Version: 4.5.0 NVIDIA 352.63`

As you can see, I have the proprietary Nvidia driver 352.63 installed. The Nvidia settings say the same.
But the driver manager (DM) insists, that I use the Nouveau driver. NV driver 352 is completely unknown to DM, only Nvidia 367 and Nvidia 340 driver get listed.

Marking the Nouveau driver as currently used is quite obviously wrong. If this would be true, how should the NVidia settings work at all? Also NVidia-prime works for me, what also indicates, that a proprietary driver is in use.

System is Mint 17.3 Cinnamon 64 bit.

AtizCooper, 2018-01-10:
Similar bug found.

- The Update manager shows:   the older version "384.90-0ubuntu0.16.04.1"   and the newer one "384.111-0ubuntu0.16.04.1" (to be upgraded to).

- nVidia-x-server-settings says:   "384.90..."  is being used.

- Synaptics says the same same thing.

- Driver Manager says:   384.111-0ubuntu0.16.04.1   is in use.
  (Sidenote: It also doesn't let me "Apply changes" for this new version "in use", obviously because it thinks its using it already.)

Attached screenshot: 
[image]

This is probably a bug in the Driver Manager displaying the wrong version of the driver in use.
System is Mint 18.3. I comment here as suggested from the support chat. Please look into it when time is available.

/ Atiz

clefebvre, 2022-09-12:
https://github.com/linuxmint/mintdrivers/commit/e98f88711910260ae0a65a6aa62c3fc5ea219f20 fixes the issue described by @AtizCooper.

Regarding the issue described by @Cos-mo however we're limited by the fact that we get the list of drivers from ubuntu-drivers-common.. if a driver goes missing from the list, we no longer see it here, whether it's installed or not, it's no longer recognized as something which relates to the device.

closed 2022-09-12 by clefebvre