linuxmint/mintdrivers #41: Failed to download repository information. Check your internet connection.

Trapizomba, 2017-08-02:
Hi!

I'm using the Linux Mint 18.2, but this problem was present at Mint 18.1 too..

I'm trying to install some system drivers but my network have proxy with authentication.

I already set the proxy and can use the update manager, apt tools, internet surfing etc

The problem is: Driver Manager

When I open it, it try to update it's informations... After some time it shows the message:

"Failed to download repository information. Check your internet connection."

Where can I set the proxy for Driver Manager?

kenden, 2018-12-12:
I had this issue also
(in Linux Mint 19.1 beta with mintdrivers 1.4.1)

mintdrivers does not read proxy information from the system

Expected: it should get the proxy information from the network settings.

This is related:
https://unix.stackexchange.com/questions/265553/driver-manager-through-proxy

clefebvre, 2022-09-12:
Duplicate https://github.com/linuxmint/mintdrivers/issues/49

closed 2022-09-12 by clefebvre