linuxmint/mintdrivers #51: Driver stuck in 'continue using manually installed' driver after switching to nouveau. others are greyed out

ijash, 2019-12-24:
```
 * Distribution : Mint 19.03
 * Cinnamon version :  4.4.5
 * 64 bit
 ```

**Issue**
I am getting this message:
[image]
Then, the driver manager shows:
[image]

the problem is, i can't get back to install the nvidia drivers. other options are greyed out.

**Steps to reproduce**
1. after long using nvidia drivers, switch back to nouveau open source driver using the driver manager.
2. restart.
3. open the driver manager, and it's stuck to the manually installed driver. Unable to switch other drivers.  

*note:
this is tested not only in a computer, but other computer as well. with the same mint distribution.*

**Expected behaviour**

Users are freely switching between drivers even if they manually install custom drivers.

**Other information**
this linux mint is upgraded version form 19.02 via update manager.  

system info:
```

System:
  Host: ijash-LM-PC Kernel: 5.0.0-37-generic x86_64 bits: 64 compiler: gcc 
  v: 7.4.0 Desktop: Cinnamon 4.4.5 wm: muffin 4.4.1 dm: LightDM 1.26.0 
  Distro: Linux Mint 19.3 Tricia base: Ubuntu 18.04 bionic 
Machine:
  Type: Desktop System: Gigabyte product: Z270X-UD3 v: N/A serial: <filter> 
  Mobo: Gigabyte model: Z270X-UD3-CF v: x.x serial: <filter> 
  UEFI: American Megatrends v: F7 date: 07/07/2017 
CPU:
  Topology: Quad Core model: Intel Core i7-7700K bits: 64 type: MT MCP 
  arch: Kaby Lake rev: 9 L2 cache: 8192 KiB 
  flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 67200 
  Speed: 800 MHz min/max: 800/4500 MHz Core speeds (MHz): 1: 800 2: 801 
  3: 800 4: 800 5: 800 6: 800 7: 800 8: 800 
Graphics:
  Device-1: NVIDIA GM206 [GeForce GTX 950] vendor: ZOTAC driver: nvidia 
  v: 435.21 bus ID: 01:00.0 chip ID: 10de:1402 
  Display: x11 server: X.Org 1.19.6 driver: nvidia 
  resolution: 1368x768~60Hz, 1920x1080~60Hz 
  OpenGL: renderer: GeForce GTX 950/PCIe/SSE2 v: 4.6.0 NVIDIA 435.21 
  direct render: Yes 
Audio:
  Device-1: Intel 200 Series PCH HD Audio vendor: Gigabyte 
  driver: snd_hda_intel v: kernel bus ID: 00:1f.3 chip ID: 8086:a2f0 
  Device-2: NVIDIA vendor: ZOTAC driver: snd_hda_intel v: kernel 
  bus ID: 01:00.1 chip ID: 10de:0fba 
  Device-3: Yamaha type: USB driver: snd-usb-audio bus ID: 1-7:3 
  chip ID: 0499:1502 
  Sound Server: ALSA v: k5.0.0-37-generic 
Network:
  Device-1: Intel Ethernet I219-V vendor: Gigabyte driver: e1000e v: 3.2.6-k 
  port: f000 bus ID: 00:1f.6 chip ID: 8086:15b8 
  IF: enp0s31f6 state: up speed: 100 Mbps duplex: full mac: <filter> 
Drives:
  Local Storage: total: 4.33 TiB used: 136.73 GiB (3.1%) 
  ID-1: /dev/nvme0n1 vendor: A-Data model: SX8200PNP size: 476.94 GiB 
  speed: 31.6 Gb/s lanes: 4 serial: <filter> rev: S0118C scheme: GPT 
  ID-2: /dev/sda vendor: Samsung model: SSD 850 EVO 250GB size: 232.89 GiB 
[... 131 lines cut ...]

(csd-power:1671): power-plugin-CRITICAL **: 01:57:55.895: abs_to_percentage: assertion 'max > min' failed
Cjs-Message: 01:57:55.923: JS LOG: [LookingGlass/info] Loaded applet sound@cinnamon.org in 52 ms
Cjs-Message: 01:57:55.944: JS LOG: [LookingGlass/info] Hiding indicator (role already handled): nm-applet
Cjs-Message: 01:57:55.947: JS LOG: [LookingGlass/info] Loaded applet network@cinnamon.org in 24 ms
Cjs-Message: 01:57:56.121: JS LOG: [LookingGlass/info] Loaded applet menu@cinnamon.org in 172 ms
Cjs-Message: 01:57:56.121: JS LOG: [LookingGlass/info] AppletManager started in 482 ms
Cjs-Message: 01:57:56.123: JS LOG: [LookingGlass/info] Cinnamon took 682 ms to start
Cjs-Message: 01:57:56.133: JS LOG: Unknown network device type, is 14
/usr/share/cinnamon-screensaver/cinnamon-screensaver-main.py:84: Warning: g_base64_encode_step: assertion 'in != NULL' failed
  css = provider.to_string()
Cjs-Message: 01:58:15.938: JS LOG: [LookingGlass/info] Adding XAppStatusIcon: mintUpdate.py (org.x.StatusIcon.PID-1974-0)
Cjs-Message: 01:58:39.082: JS LOG: [LookingGlass/info] Adding XAppStatusIcon: mintreport (org.x.StatusIcon.PID-2125-0)
Cjs-Message: 03:34:14.640: JS LOG: [LookingGlass/info] Adding indicator: transmission
** Message: 03:34:38.838: Unable to select area using GNOME Shell's builtin screenshot interface, resorting to fallback X11.
Gtk-Message: 03:35:29.688: GtkDialog mapped without a transient parent. This is discouraged.
Cinnamon warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x3a00003 (Authentica)
** Message: 03:35:44.109: Unable to select area using GNOME Shell's builtin screenshot interface, resorting to fallback X11.
Cinnamon warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x4200085 (syslog [Re)
```

I have found this conversation on the [forum](https://forums.linuxmint.com/viewtopic.php?t=293438)
but I have not tried the solution when this report was posted.

any further assistance or instruction for this *(probably)*bug i .

vpalos, 2020-09-09:
Did you find a solution for this issue? I'm experiencing exactly the same behavior on a recently (freshly) installed Ubuntu 20.04.

vpalos, 2020-09-09:
For anyone who might encounter this thread, I've solved my problem by running this command:
```sh
sudo ubuntu-drivers autoinstall
```
After running (and installing a bunch of packages) the selected installed driver became the recommended one (run `ubuntu-drivers devices` to find out what that is). Don't forget to reboot afterwards.

realfoodman, 2020-11-22:
These two things worked for me on Ubuntu 12.10. Thank you! After running the commands and rebooting, the screen was at low resolution, so I switched to the next-newest driver (which is what Ubuntu had selected by default after being installed) and rebooted again. Smooth sailing since then.

mtwebster, 2020-11-25:
Is this still occurring in the current version of the driver manager? (1.5.2).

mghoffmann, 2020-12-04:
Yes, it is. I'm trying to solve it right now. For me it started after installing NVidia Cuda following the instructions at https://developer.nvidia.com/cuda-downloads.

kmanwar89, 2022-06-02:
> For anyone who might encounter this thread, I've solved my problem by running this command:
> 
> ```shell
> sudo ubuntu-drivers autoinstall
> ```
> 
> After running (and installing a bunch of packages) the selected installed driver became the recommended one (run `ubuntu-drivers devices` to find out what that is). Don't forget to reboot afterwards.

Since this issue is still open...I faced this exact issue on a fresh install of Ubuntu 22.04. Running the command from @vpalos worked for me as well! Thank you for posting the solution.

jabowery, 2023-02-24:
```
# ubuntu-drivers autoinstall
Traceback (most recent call last):
  File "/usr/bin/ubuntu-drivers", line 513, in <module>
    greet()
  File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/usr/bin/ubuntu-drivers", line 432, in autoinstall
    command_install(config)
  File "/usr/bin/ubuntu-drivers", line 187, in command_install
    UbuntuDrivers.detect.nvidia_desktop_pre_installation_hook(to_install)
  File "/usr/lib/python3/dist-packages/UbuntuDrivers/detect.py", line 839, in nvidia_desktop_pre_installation_hook
    with_nvidia_kms = version >= 470
UnboundLocalError: local variable 'version' referenced before assignment

```

DolceParadise, 2023-02-25:
> ```
> # ubuntu-drivers autoinstall
> Traceback (most recent call last):
>   File "/usr/bin/ubuntu-drivers", line 513, in <module>
>     greet()
>   File "/usr/lib/python3/dist-packages/click/core.py", line [...]

I am getting the same error on `sudo ubuntu-drivers autoinstall`
Is there a workaround for this?

DjSapsan, 2023-03-16:
Solution:
sudo apt purge *nvidia*
sudo apt autoremove
sudo apt install --reinstall libgl1-mesa-glx libgl1-mesa-dri xserver-xorg-core xserver-xorg-video-nouveau libdrm-nouveau2

then logout and in

suddenly everything is selectable!

Committing, 2023-03-27:
> @DolceParadise replace the line 853 with `version = int(package_name.split('-')[-2])` in `/usr/lib/python3/dist-packages/UbuntuDrivers/detect.py`

Thanks, it is line 835 now.

edison23, 2023-04-09:
Installing the `nvidia-driver-525` explicitly using APT fixed it for me: https://askubuntu.com/a/1463002/333914

Aakash-chouksey, 2023-05-12:
> @DolceParadise replace the line 853 with `version = int(package_name.split('-')[-2])` in `/usr/lib/python3/dist-packages/UbuntuDrivers/detect.py`

worked for me. I was facing the issue of black screen after boot so i change the driver to nouveau driver from the during the bootup process after that my drivers got locked to "continue using manually installed driver " in the ubuntu addition drivers 

after  making this changes and rebooting i was able to change my driver from addition driver setting but it is still facing the blackout screen problem

I have one more question should i be using the nvidia drivers open(porprietary,tested ) ones or just the one proprietary option 
thank you in advance