linuxmint/mintdrivers #58: [Feature Request] Warning regarding secureboot

kelebek333, 2020-09-14:
Hi,

Could you add a warn (or tooltip) related for disabiling secureboot if it enabled.

Best regards

clefebvre, 2020-12-12:
I agree, we should do that.

We need to rethink the UI slightly as well. It's not designed to show information right now, it goes straight into refreshing mode. I don't think we'll get this ready for BETA or maybe not even for STABLE, but we'll definitely add it before 20.2.

LinuxOnTheDesktop, 2022-02-03:
> we'll definitely add it before 20.2

So . .

clefebvre, 2022-09-15:
hi @kelebek333,

Please explain why this is needed.

I agreed on principle but while working on this we found a limitation between mintdrivers and nvidia drivers in secureboot we just managed to overcome, so I'm wondering if this warning is still needed or not.

I'll try to explain the situation the best I can. This all applies to Mint 21 (probably also to earlier releases).

- The nvidia driver packages trigger a call to dkms in their postinst (i.e. right after they're installed dkms runs to recompile the NVIDIA kernel module)
- dkms compiles a new NVIDIA kernel module and checks if secureboot is ON or OFF.
- If secureboot is ON it also signs the module, asks the user for a password and tells him how to authorize the module in his BIOS settings.
- The way it asks and shows all this information is by using debconf.

Now here's the bug we found:

- mintdrivers runs as root and uses packagekit to install the drivers.
- packagekit supports debconf but this support only works when it's run as user, not as root.

So with secureboot ON, what happens is this:

- mintdrivers runs as root and asks packagekit to install nvidia drivers
- dkms is triggered and asks debconf to show information to the user
- debconf can't find the user socket and fails/hangs
- mintdrivers consequently hangs for a while and crashes
- the drivers aren't fully installed and aren't authorized in secureboot

clefebvre, 2022-09-15:
So with this in mind we started to rework mintdrivers for it to run in user mode. This has other advantages such as the fact that it will inherit user proxy settings. I just tested this new version we're working on in a secureboot environment and it succeeds in installing nvidia drivers. It asks the user for a password as if you installed the drivers yourself.

clefebvre, 2022-09-15:
Confirmed after testing on an NVIDIA optimus laptop using secureBoot:

- Installed Mint 21 Cinnamon
- Swapped mintdrivers for the new version (the one which runs as user)
- Selected nvidia-driver-515
- A password is asked to sign the module and instructions are given (this didn't appear before)
- On reboot the laptop presents a BIOS screen to enroll the MOK
- Inserted password and enrolled the MOK as instructed
- The driver works fine, it defaults to NVIDIA on-demand
- Cinnamon menu's Run with NVIDIA GPU works fine
- Nvidia-prime successfully switches to NVIDIA performance upon logout
- Upgrading the kernel with mintupdate works fine, no debconf triggered, driver still works.

closed 2022-11-19 by kelebek333