linuxmint/mintdrivers #60: mintdrivers 1.5.1 stuck on "Looking for hardware drivers ...", never finishes

KHeilen, 2020-11-07:
Reference:
Mint forum thread: [Driver Manager hangs after mintdrivers update (LM 20 XFCE)](https://forums.linuxmint.com/viewtopic.php?f=47&t=334467)

Affected systems:
- [Linux Mint 20 xfce system](https://forums.linuxmint.com/viewtopic.php?p=1915802#p1915802)
- [Linux Mint 19.3 xfce 32-bit system](https://forums.linuxmint.com/viewtopic.php?p=1916133#p1916133)

Symptoms in both cases:
> I am running LM 20 XFCE.
This morning I had two updates; mintdrivers and mintsystem.
Now I discovered that when I run Driver Manager it hangs with an animated rotating "working" spinner and the message:
`Looking for hardware drivers...`
I tried it several times and the same thing happens. It hangs with 0% CPU usage until I close the window.

> This morning, I received mintdrivers 1.5.1 on a third machine, LM 19.3 xfce 32-bit. Launched mintdrivers and saw the reported behaviour: `Looking for hardware drivers...` Permitted mintdrivers to run for 10 minutes. Then clicked on the [x] close button.
On a second try, waited for 15 minutes hoping mintdrivers would finish `Looking for hardware drivers...`, then closed mintdrivers.

System information for the [LM 19.3 xfce 32-bit system](https://termbin.com/4yu3)
I have not found any errors in any error log or system log on this system, which would explain why mintdrivers never finishes `Looking for hardware drivers...` 

Note:
2 other LM 19.3 Cinnamon 64-bit systems had received mintdrivers 1.5.1 as well. 
On these 2 machines, mintdrivers completed `Looking for hardware drivers...` after 30 seconds roughly and displayed the found drivers as expected.

In case more details need to be provided, please, let me know.

[LM19.3xfce_inxi.txt](https://github.com/linuxmint/mintdrivers/files/5505523/LM19.3xfce_inxi.txt)

KHeilen, 2020-11-08:
In the [referenced thread](https://forums.linuxmint.com/viewtopic.php?p=1916567#p1916567) , another affected user shared the terminal output, which was displayed, when he executed mintdrivers in a terminal window on LM 19.3 Cinnamon:

> Kernel: 5.4.0-52-generic x86_64 bits: 64 
Desktop: Cinnamon 4.4.8 Distro: Linux Mint 19.3 Tricia 
Starting driver manager in a terminal gives:

```
tim@tim:~$ driver-manager
tim@tim:~$ Post-install mode detected
Checking connectivity or live media...
Internet connection detected
Ignoring server NVIDIA driver:  nvidia-driver-450-server
Ignoring server NVIDIA driver:  nvidia-driver-418-server
Ignoring server NVIDIA driver:  nvidia-driver-440-server
Traceback (most recent call last):
  File "/usr/lib/linuxmint/mintdrivers/mintdrivers.py", line 585, in show_drivers
    if driver.startswith("nvidia-driver") and driver.endswith("-server"):
AttributeError: 'bool' object has no attribute 'startswith'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 154, in apport_excepthook
    os.O_WRONLY | os.O_CREAT | os.O_EXCL, 0o640), 'wb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_usr_lib_linuxmint_mintdrivers_mintdrivers.py.1000.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/lib/linuxmint/mintdrivers/mintdrivers.py", line 585, in show_drivers
    if driver.startswith("nvidia-driver") and driver.endswith("-server"):
AttributeError: 'bool' object has no attribute 'startswith'

```

KHeilen, 2020-11-08:
By now executed mintdrivers on my LM 19.3 xfce 32-bit machine in the terminal window twice as well and collected the terminal output.

On first try today, I waited for mintdrivers for 20 min., just to be reasonably sure that it will not finish.
Actually, after 5 min. the terminal output stopped changing (no more new output)
On the second try therefore, mintdrivers was closed after 6 min.
The second try was done after having created the missing directory /var/crash. (Cf. the terminal output)
But as mintdrivers starts under the user account, it is not permitted to write to /var/crash.

```
First try today: 

(11:36)
karl@paulchen:~$ Post-install mode detected
Checking connectivity or live media...
Internet connection detected
Traceback (most recent call last):
  File "/usr/lib/linuxmint/mintdrivers/mintdrivers.py", line 585, in show_drivers
    if driver.startswith("nvidia-driver") and driver.endswith("-server"):
AttributeError: 'bool' object has no attribute 'startswith'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 154, in apport_excepthook
    os.O_WRONLY | os.O_CREAT | os.O_EXCL, 0o640), 'wb') as f:
FileNotFoundError: [Errno 2] Datei oder Verzeichnis nicht gefunden: '/var/crash/_usr_lib_linuxmint_mintdrivers_mintdrivers.py.1000.crash'
Original exception was:
Traceback (most recent call last):
  File "/usr/lib/linuxmint/mintdrivers/mintdrivers.py", line 585, in show_drivers
    if driver.startswith("nvidia-driver") and driver.endswith("-server"):
AttributeError: 'bool' object has no attribute 'startswith'
(11:40) 

11:36 => 11:37 repos queried without problems
11:36 => 11:40 cpu load mintdrivers 43%
11:40          cpu load mintdrivers down to 5%
11:42          cpu load mintdrivers 0% => 5-6%
11:56 (closed mintdrivers application window [X])

sudo mkdir /var/crash

Second try:

So 8. Nov 11:58:55 CET 2020
karl@paulchen:~$ Post-install mode detected
Checking connectivity or live media...
Internet connection detected
Traceback (most recent call last):
  File "/usr/lib/linuxmint/mintdrivers/mintdrivers.py", line 585, in show_drivers
    if driver.startswith("nvidia-driver") and driver.endswith("-server"):
AttributeError: 'bool' object has no attribute 'startswith'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 154, in apport_excepthook
    os.O_WRONLY | os.O_CREAT | os.O_EXCL, 0o640), 'wb') as f:
PermissionError: [Errno 13] Keine Berechtigung: '/var/crash/_usr_lib_linuxmint_mintdrivers_mintdrivers.py.1000.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/lib/linuxmint/mintdrivers/mintdrivers.py", line 585, in show_drivers
    if driver.startswith("nvidia-driver") and driver.endswith("-server"):
AttributeError: 'bool' object has no attribute 'startswith'

(12:02)

(12:04) closed mintdrivers

BiNh0X, 2020-11-08:
I'm too.

I have Linux Mint 19.3 and after the last update of mintdrivers to version 1.5.1 installed on November 6th my Driver Manager also stopped working. "Looking for hardware driver..." forever.

Post-install mode detected
Checking connectivity or live media...
Internet connection detected
Traceback (most recent call last):
  File "/usr/lib/linuxmint/mintdrivers/mintdrivers.py", line 585, in show_drivers
    if driver.startswith("nvidia-driver") and driver.endswith("-server"):
AttributeError: 'bool' object has no attribute 'startswith'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 154, in apport_excepthook
    os.O_WRONLY | os.O_CREAT | os.O_EXCL, 0o640), 'wb') as f:
FileNotFoundError: [Errno 2] Arquivo ou diretório inexistente: '/var/crash/_usr_lib_linuxmint_mintdrivers_mintdrivers.py.0.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/lib/linuxmint/mintdrivers/mintdrivers.py", line 585, in show_drivers
    if driver.startswith("nvidia-driver") and driver.endswith("-server"):
AttributeError: 'bool' object has no attribute 'startswith'

mtwebster, 2020-11-08:
Hi, unfortunately I can't reproduce this at the moment, can I get some more info from everyone where possible:

- Open "System Reports" (mintreport), go to the system information tab, and click upload. Wait for a notification with the link and paste it here.
- Run the following:
```
apt list ubuntu-system-adjustments
python3 --version
which python3
```
and paste the output here.

Thanks

BiNh0X, 2020-11-08:
1. System Reports (mintreport) > system information
```
System:    Host: binhocastro-NE56R Kernel: 5.4.0-52-generic x86_64 bits: 64 compiler: gcc v: 7.5.0 
           Desktop: Cinnamon 4.4.8 wm: muffin dm: LightDM Distro: Linux Mint 19.3 Tricia 
           base: Ubuntu 18.04 bionic 
Machine:   Type: Laptop System: Gateway product: NE56R v: V1.13 serial: <filter> Chassis: type: 10 
           v: 1.13 serial: <filter> 
           Mobo: Gateway model: EG50_HC_HR v: Type2 - Board Version serial: <filter> 
           UEFI [Legacy]: Gateway v: 1.13 date: 10/09/2012 
Battery:   ID-1: BAT1 charge: 18.4 Wh condition: 18.4/47.5 Wh (39%) volts: 12.2/10.8 
           model: SANYO Li_Ion_4000mA serial: <filter> status: Full 
CPU:       Topology: Dual Core model: Intel Pentium 2020M bits: 64 type: MCP arch: Ivy Bridge 
           rev: 9 L2 cache: 2048 KiB 
           flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 9577 
           Speed: 2394 MHz min/max: 1200/2400 MHz Core speeds (MHz): 1: 2394 2: 2394 
Graphics:  Device-1: Intel 3rd Gen Core processor Graphics vendor: Acer Incorporated ALI 
           driver: i915 v: kernel bus ID: 00:02.0 chip ID: 8086:0156 
           Display: x11 server: X.Org 1.20.8 driver: modesetting unloaded: fbdev,vesa 
           resolution: 1920x1080~60Hz 
           OpenGL: renderer: Mesa DRI Intel HD Graphics 2500 (IVB GT1) v: 4.2 Mesa 20.0.8 
           compat-v: 3.0 direct render: Yes 
Audio:     Device-1: Intel 7 Series/C216 Family High Definition Audio 
           vendor: Acer Incorporated ALI driver: snd_hda_intel v: kernel bus ID: 00:1b.0 
           chip ID: 8086:1e20 
           Sound Server: ALSA v: k5.4.0-52-generic 
Network:   Device-1: Broadcom and subsidiaries NetLink BCM57785 Gigabit Ethernet PCIe 
           vendor: Acer Incorporated ALI driver: tg3 v: 3.137 port: 2040 bus ID: 02:00.0 
           chip ID: 14e4:16b5 
           IF: enp2s0f0 state: down mac: <filter> 
           Device-2: Broadcom and subsidiaries BCM4313 802.11bgn Wireless Network Adapter 
           vendor: Hewlett-Packard driver: wl v: kernel port: 2040 bus ID: 03:00.0 
           chip ID: 14e4:4727 
           IF: wlp3s0 state: up mac: <filter> 
           IF-ID-1: vmnet1 state: unknown speed: N/A duplex: N/A mac: <filter> 
           IF-ID-2: vmnet8 state: unknown speed: N/A duplex: N/A mac: <filter> 
Drives:    Local Storage: total: 689.33 GiB used: 479.31 GiB (69.5%) 
           ID-1: /dev/sda vendor: Kingston model: SA400S37240G size: 223.57 GiB speed: 6.0 Gb/s 
           serial: <filter> 
           ID-2: /dev/sdb type: USB vendor: Seagate model: ST500LM012 HN-M500MBB size: 465.76 GiB 
           serial: <filter> 
Partition: ID-1: / size: 95.62 GiB used: 29.82 GiB (31.2%) fs: ext4 dev: /dev/sda5 
USB:       Hub: 1-0:1 info: Full speed (or root) Hub ports: 2 rev: 2.0 chip ID: 1d6b:0002 
           Hub: 1-1:2 info: Intel Integrated Rate Matching Hub ports: 4 rev: 2.0 
           chip ID: 8087:0024 
           Device-1: 1-1.1:3 info: Broadcom BCM2070 Bluetooth 2.1 + EDR type: Bluetooth 
           driver: btusb rev: 2.0 chip ID: 0a5c:21b4 
           Device-2: 1-1.3:4 info: Chicony type: Video driver: uvcvideo rev: 2.0 
           chip ID: 04f2:b337 
           Hub: 2-0:1 info: Full speed (or root) Hub ports: 2 rev: 2.0 chip ID: 1d6b:0002 
           Hub: 2-1:2 info: Intel Integrated Rate Matching Hub ports: 4 rev: 2.0 
           chip ID: 8087:0024 
           Device-3: 2-1.2:4 info: JMicron / JMicron USA JM20329 SATA Bridge type: Mass Storage 
           driver: usb-storage rev: 2.0 chip ID: 152d:2329 
Sensors:   System Temperatures: cpu: 57.0 C mobo: N/A 
           Fan Speeds (RPM): N/A 
Repos:     Active apt repos in: /etc/apt/sources.list 
           1: deb https: //typora.io/linux ./
           Active apt repos in: /etc/apt/sources.list.d/additional-repositories.list 
           1: deb https: //download.onlyoffice.com/repo/debian squeeze main
           Active apt repos in: /etc/apt/sources.list.d/balena-etcher.list 
           1: deb https: //deb.etcher.io stable etcher
           Active apt repos in: /etc/apt/sources.list.d/google-chrome.list 
           1: deb [arch=amd64] http: //dl.google.com/linux/chrome/deb/ stable main
           Active apt repos in: /etc/apt/sources.list.d/headset.list 
           1: deb [arch=amd64] http: //headsetapp.co/headset-electron/debian stable non-free
           Active apt repos in: /etc/apt/sources.list.d/megasync.list 
           1: deb https: //mega.nz/linux/MEGAsync/xUbuntu_18.04/ ./
           Active apt repos in: /etc/apt/sources.list.d/nilarimogard-webupd8-bionic.list 
           1: deb http: //ppa.launchpad.net/nilarimogard/webupd8/ubuntu bionic main
           Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list 
[... 7 lines cut ...]
           1: deb https: //deb.opera.com/opera-stable/ stable non-free #Opera Browser (final releases)
           Active apt repos in: /etc/apt/sources.list.d/rvm-smplayer-bionic.list 
           1: deb http: //ppa.launchpad.net/rvm/smplayer/ubuntu bionic main
           Active apt repos in: /etc/apt/sources.list.d/skype-stable.list 
           1: deb [arch=amd64] https: //repo.skype.com/deb stable main
           Active apt repos in: /etc/apt/sources.list.d/spotify.list 
           1: deb http: //repository.spotify.com stable non-free
           Active apt repos in: /etc/apt/sources.list.d/teamviewer.list 
[... cut ...]
Info:      Processes: 254 Uptime: 1h 06m Memory: 7.62 GiB used: 2.48 GiB (32.5%) Init: systemd 
           v: 237 runlevel: 5 Compilers: gcc: 7.5.0 alt: 7 Client: Unknown python3.6 client 
           inxi: 3.0.32
``` 
```
apt list ubuntu-system-adjustments ubuntu-drivers-common
Listing... Pronto
ubuntu-drivers-common/bionic-updates,now 1:0.5.2.5 amd64 [installed]
ubuntu-system-adjustments/tricia,tricia,now 2019.12.09.2 all [installed]
binho-castro@binhocastro-NE56R:~$ python3 --version
Python 3.6.9
binho-castro@binhocastro-NE56R:~$ which python3
/usr/bin/python3
```

KHeilen, 2020-11-08:
Good evening, mtwebster.

Thanks a lot for your post. Please, find the requested pieces of information below.

```$ inxi -Sx3z
System:    Kernel: 4.15.0-122-generic i686 bits: 32 compiler: gcc v: 7.5.0 Desktop: Xfce 4.14.1 tk: Gtk 3.22.30 
           info: xfce4-panel wm: xfwm4 dm: LightDM 1.26.0 Distro: Linux Mint 19.3 Tricia base: Ubuntu 18.04 bionic
```

System Reports => System Information (inxi report): https://termbin.com/4yu3

```
System:    Kernel: 4.15.0-122-generic i686 bits: 32 compiler: gcc v: 7.5.0 Desktop: Xfce 4.14.1 
           tk: Gtk 3.22.30 wm: xfwm4 dm: LightDM Distro: Linux Mint 19.3 Tricia 
           base: Ubuntu 18.04 bionic 
Machine:   Type: Laptop System: Acer product: AOD260 v: V1.08_Vodafone serial: <filter> 
           Mobo: Acer model: AOD260 v: V1.08_Vodafone serial: <filter> BIOS: Acer v: 1.08_Vodafone 
           date: 06/29/2010 
Battery:   ID-1: BAT0 charge: 12.4 Wh condition: 12.4/48.8 Wh (25%) volts: 12.5/11.1 
           model: SANYO AL10B31 serial: N/A status: Full 
CPU:       Info: Single Core model: Intel Atom N450 bits: 64 type: MT arch: Bonnell rev: A 
           L2 cache: 512 KiB 
           flags: lm nx pae sse sse2 sse3 ssse3 bogomips: 6649 
           Speed: 1662 MHz min/max: 1000/1666 MHz Core speeds (MHz): 1: 1637 2: 1578 
Graphics:  Device-1: Intel Atom Processor D4xx/D5xx/N4xx/N5xx Integrated Graphics 
           vendor: Acer Incorporated ALI driver: i915 v: kernel bus ID: 00:02.0 chip ID: 8086:a011 
           Device-2: ALi Gateway Webcam type: USB driver: uvcvideo bus ID: 1-4:3 
           chip ID: 0402:9665 
           Display: x11 server: X.Org 1.19.6 compositor: compton driver: intel 
           unloaded: fbdev,modesetting,vesa resolution: 1024x600~60Hz s-dpi: 96 
           OpenGL: renderer: Mesa DRI Intel Pineview M x86/MMX/SSE2 v: 1.4 Mesa 20.0.8 
           direct render: Yes 
Audio:     Device-1: Intel NM10/ICH7 Family High Definition Audio vendor: Acer Incorporated ALI 
           driver: snd_hda_intel v: kernel bus ID: 00:1b.0 chip ID: 8086:27d8 
           Sound Server: ALSA v: k4.15.0-122-generic 
Network:   Device-1: Qualcomm Atheros AR8132 Fast Ethernet vendor: Acer Incorporated ALI 
           driver: atl1c v: 1.0.1.1-NAPI port: 5000 bus ID: 01:00.0 chip ID: 1969:1062 
           IF: enp1s0 state: down mac: <filter> 
           Device-2: Broadcom and subsidiaries BCM4313 802.11bgn Wireless Network Adapter 
           driver: wl v: kernel port: 5000 bus ID: 02:00.0 chip ID: 14e4:4727 
           IF: wlp2s0 state: up mac: <filter> 
Drives:    Local Storage: total: 232.89 GiB used: 13.10 GiB (5.6%) 
           ID-1: /dev/sda vendor: Toshiba model: MK2565GSX size: 232.89 GiB speed: 3.0 Gb/s 
           serial: <filter> 
Partition: ID-1: / size: 69.01 GiB used: 12.91 GiB (18.7%) fs: ext4 dev: /dev/sda5 
Swap:      ID-1: swap-1 type: partition size: 2.00 GiB used: 192.8 MiB (9.4%) priority: -2 
           dev: /dev/sda3 
USB:       Hub: 1-0:1 info: Full speed (or root) Hub ports: 8 rev: 2.0 chip ID: 1d6b:0002 
           Device-1: 1-4:3 info: ALi Gateway Webcam type: Video driver: uvcvideo rev: 2.0 
           chip ID: 0402:9665 
           Hub: 2-0:1 info: Full speed (or root) Hub ports: 2 rev: 1.1 chip ID: 1d6b:0001 
           Device-1: 2-2:2 info: Primax 0Y357C PMX-MMOCZUL (B) [Dell Laser Mouse] type: Mouse 
           driver: hid-generic,usbhid rev: 2.0 chip ID: 0461:4d51 
           Hub: 3-0:1 info: Full speed (or root) Hub ports: 2 rev: 1.1 chip ID: 1d6b:0001 
           Hub: 4-0:1 info: Full speed (or root) Hub ports: 2 rev: 1.1 chip ID: 1d6b:0001 
           Hub: 5-0:1 info: Full speed (or root) Hub ports: 2 rev: 1.1 chip ID: 1d6b:0001 
Sensors:   System Temperatures: cpu: 45.0 C mobo: N/A 
           Fan Speeds (RPM): N/A 
Repos:     Packages: 2169 apt: 2165 flatpak: 4 
           No active apt repos in: /etc/apt/sources.list 
           Active apt repos in: /etc/apt/sources.list.d/gezakovacs-ppa-bionic.list 
           1: deb http: //ppa.launchpad.net/gezakovacs/ppa/ubuntu bionic main
           Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list 
           1: deb http: //packages.linuxmint.com tricia main upstream import backport #id:linuxmint_main
           2: deb http: //de.archive.ubuntu.com/ubuntu bionic main restricted universe multiverse
           3: deb http: //de.archive.ubuntu.com/ubuntu bionic-updates main restricted universe multiverse
           4: deb http: //de.archive.ubuntu.com/ubuntu bionic-backports main restricted universe multiverse
           5: deb http: //security.ubuntu.com/ubuntu/ bionic-security main restricted universe multiverse
           6: deb http: //archive.canonical.com/ubuntu/ bionic partner
           Active apt repos in: /etc/apt/sources.list.d/softmaker.list 
[... 5 lines cut ...]
           Active apt repos in: /etc/apt/sources.list.d/unit193-inxi-bionic.list 
           1: deb http: //ppa.launchpad.net/unit193/inxi/ubuntu bionic main
           Active apt repos in: /etc/apt/sources.list.d/vivaldi.list 
           1: deb http: //repo.vivaldi.com/stable/deb/ stable main
Info:      Processes: 176 Uptime: 1h 14m Memory: 987.1 MiB used: 573.2 MiB (58.1%) Init: systemd 
           v: 237 runlevel: 5 Compilers: gcc: 7.5.0 alt: 7 Client: Unknown python3.6 client 
           inxi: 3.1.08 
```

```
$ apt list ubuntu-system-adjustments ubuntu-drivers-common
Listing... Done
ubuntu-drivers-common/bionic-updates,now 1:0.5.2.5 i386 [installed]
ubuntu-system-adjustments/tricia,now 2019.12.09.2 all [installed]

$ python3 --version
Python 3.6.9

$ which python3
/usr/bin/python3

$ ls -l /usr/bin/python3
lrwxrwxrwx 1 root root 9 Jul 31  2019 /usr/bin/python3 -> python3.6

```

AlexB7, 2020-11-09:
I am having the same issue with Mint 20 Cinnamon, freshly installed and then fully updated.  I was able to get in to driver manager after the installation to install a wireless driver.  After updating I went back to install the NVidia driver and the problem started.

```
$ mintdrivers
$ Post-install mode detected
Checking connectivity or live media...
Internet connection detected
Traceback (most recent call last):
  File "/usr/lib/linuxmint/mintdrivers/mintdrivers.py", line 585, in show_drivers
    if driver.startswith("nvidia-driver") and driver.endswith("-server"):
AttributeError: 'bool' object has no attribute 'startswith'
```

```
System:    Kernel: 5.4.0-52-generic x86_64 bits: 64 compiler: gcc v: 9.3.0 Desktop: Cinnamon 4.6.7 
           wm: muffin dm: LightDM Distro: Linux Mint 20 Ulyana base: Ubuntu 20.04 focal 
Graphics:  Device-1: NVIDIA MCP89 [GeForce 320M] vendor: Apple driver: nouveau v: kernel 
           bus ID: 04:00.0 chip ID: 10de:08a0 
           Display: x11 server: X.Org 1.20.8 driver: modesetting unloaded: fbdev,vesa 
           resolution: 1280x800~60Hz 
           OpenGL: renderer: NVAF v: 3.3 Mesa 20.0.8 direct render: Yes 
Audio:     Device-1: NVIDIA MCP89 High Definition Audio driver: snd_hda_intel v: kernel 
           bus ID: 00:08.0 chip ID: 10de:0d94 
           Sound Server: ALSA v: k5.4.0-52-generic 
Repos:     No active apt repos in: /etc/apt/sources.list 
           No active apt repos in: /etc/apt/sources.list.d/brave-browser-release.list 
           Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list 
           1: deb http: //muug.ca/mirror/linuxmint/packages ulyana main upstream import backport
           2: deb http: //mirror.ca-tr.kamatera.com/ubuntu focal main restricted universe multiverse
           3: deb http: //mirror.ca-tr.kamatera.com/ubuntu focal-updates main restricted universe multiverse
           4: deb http: //mirror.ca-tr.kamatera.com/ubuntu focal-backports main restricted universe multiverse
           5: deb http: //security.ubuntu.com/ubuntu/ focal-security main restricted universe multiverse
           6: deb http: //archive.canonical.com/ubuntu/ focal partner
```

```
apt list ubuntu-system-adjustments ubuntu-drivers-common mintdrivers
Listing... Done
ubuntu-drivers-common/focal-updates,now 1:0.8.4~0.20.04.3 amd64 [installed]
ubuntu-system-adjustments/ulyana,ulyana,now 2020.07.29 all [installed]
mintdrivers/ulyana,ulyana,now 1.5.1 all [installed]
```

```
# python3 --version
Python 3.8.5
$ which python3
/usr/bin/python3
```

AlexB7, 2020-11-09:
A *temporary workaround* is to apply this patch which adds a check before calling startswith() and endswith().

```
--- /usr/lib/linuxmint/mintdrivers/mintdrivers.py.org	2020-11-08 19:29:51.508519974 -0500
+++ /usr/lib/linuxmint/mintdrivers/mintdrivers.py	2020-11-08 19:29:01.817081387 -0500
@@ -582,9 +582,10 @@
                 # define the order of introspection
                 for section in ('recommended', 'alternative', 'manually_installed', 'no_driver'):
                     for driver in sorted(drivers[section], key=lambda x: self.sort_string(drivers[section], x)):
-                        if driver.startswith("nvidia-driver") and driver.endswith("-server"):
-                            print("Ignoring server NVIDIA driver: ", driver)
-                            continue
+                        if(isinstance(drivers,str)): 
+                            if driver.startswith("nvidia-driver") and driver.endswith("-server"):
+                                print("Ignoring server NVIDIA driver: ", driver)
+                                continue
                         radio_button = Gtk.RadioButton.new(None)
                         label = Gtk.Label()
                         label.set_markup(drivers[section][driver]['description'])
```
After this change I am able to get in to the drivers screen.

anysystemroot, 2020-11-09:
Thank you, AlexB7.
i apply your patch, but i have new problem:
"pk-client-error-quark: Error while installing package: installed bcmwl-kernel-source package post-installation script subprocess returned error exit status 10 (313)"
ps: after try to install broadcom proprietary driver =(

mtwebster, 2020-11-09:
i see some of you don't have discrete graphics, are all of you using mintdrivers for a wireless device?

Can you report `lshw -class network`?

anysystemroot, 2020-11-09:
> lshw
https://pastebin.com/VGSqUQdr

commit 27939e8a (2020-11-09): Fix #60

The driver variable can be a boolean in certain cases, not a string.
closed 2020-11-09 by clefebvre