linuxmint/mintdrivers #61: driver-manager: bcmwl-kernel-source (for BCM43142)

anysystemroot, 2020-11-09:
i have the error:
"pk-client-error-quark: Error while installing package: installed bcmwl-kernel-source package post-installation script subprocess returned error exit status 10 (313)"

anysystemroot, 2020-11-09:
stdout:
...
Loading new bcmwl-6.30.223.271+bdcom DKMS files...
Building for 5.8.0-25-generic
Building for architecture x86_64
Building initial module for 5.8.0-25-generic
Error! Bad return status for module build on kernel: 5.8.0-25-generic (x86_64)
Consult /var/lib/dkms/bcmwl/6.30.223.271+bdcom/build/make.log for more information.
dpkg: ошибка при обработке пакета bcmwl-kernel-source (--configure):
 installed bcmwl-kernel-source package post-installation script subprocess returned error exit status 10
...

mtwebster, 2020-11-09:
Is this with mintdrivers 1.5.2 installed? [this](https://github.com/linuxmint/mintdrivers/issues/60) issue should be fixed now - I noted you had this issue using a suggested patch in that thread.

Please try this with the new official version if you haven't yet.

anysystemroot, 2020-11-09:
yes, but i install kernel-5.8.0.25, maybe it's problem.. i mean.

clefebvre, 2020-11-09:
can you reproduce the issue with the LTS kernel?

anysystemroot, 2020-11-09:
I'll try later (probably tomorrow)
I can't afford a reboot, yet =)

closed 2020-11-12 by clefebvre