linuxmint/mintdrivers #65: Issue with kernel 5.12 and nvidia?

larsmartin4, 2021-05-02:
Hi i have issue with driver-manager (mindrivers) with 20.1 cinamon.

Its works nice with ubuntu 21.04 so can you upgrade you program to support kernel 5.12?

Post-install mode detected
Checking Internet connectivity...
  --> Internet connection detected.
WARNING:root:_pkg_get_support nvidia-driver-390: package has invalid Support Legacyheader, cannot determine support level
Ignoring server NVIDIA driver:  nvidia-driver-418-server
Ignoring server NVIDIA driver:  nvidia-driver-450-server
Ignoring server NVIDIA driver:  nvidia-driver-460-server

Its missing supported kernel ubuntu 21.04 have:
software-properties-gtk --open-tab=4
WARNING:root:_pkg_get_support nvidia-driver-390: package has invalid Support Legacyheader, cannot determine support level
dpkg: warning: version 'unsigned-5.1.0-050100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.0-051100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.1.0-050100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.11-051111' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.0-051100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.12.0-051200' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.11-051111' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.9.12-050912' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.12.0-051200' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.1.0-050100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.0-051100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.1.0-050100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.11-051111' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.0-051100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.12.0-051200' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.11-051111' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.9.12-050912' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.12.0-051200' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.1.0-050100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.0-051100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.1.0-050100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.11-051111' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.0-051100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.12.0-051200' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.11-051111' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.9.12-050912' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.12.0-051200' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.1.0-050100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.0-051100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.1.0-050100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.11-051111' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.0-051100' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.12.0-051200' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.11.11-051111' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.9.12-050912' has bad syntax: version number does not start with a digit
dpkg: warning: version 'unsigned-5.12.0-051200' has bad syntax: version number does not start with a digit

larsmartin4, 2022-03-19:
did anyone know? kernel and nvidia did not support or need later ubuntu?

larsmartin4, 2022-03-19:
[image]
[image]
[image]

clefebvre, 2022-09-12:
I'm sorry, this isn't related to the driver manager itself. All it does is install/remove the drivers. Whether the drivers support your kernel series or not is handled upstream in Ubuntu and/or NVIDIA even.

closed 2022-09-12 by clefebvre