linuxmint/mintdrivers #105: Can't load mintdrivers_1.8.1_all.deb Package ubuntu-drivers-common is not installed.

ichibrosan, 2025-03-10:
sudo dpkg -i mintdrivers_1.8.1_all.deb 
Selecting previously unselected package mintdrivers.
(Reading database ... 630198 files and directories currently installed.)
Preparing to unpack mintdrivers_1.8.1_all.deb ...
Unpacking mintdrivers (1.8.1) ...
dpkg: dependency problems prevent configuration of mintdrivers:
 mintdrivers depends on ubuntu-drivers-common; however:
  Package ubuntu-drivers-common is not installed.

dpkg: error processing package mintdrivers (--install):
 dependency problems - leaving unconfigured
Processing triggers for gnome-menus (3.36.0-1.1) ...
Processing triggers for mate-menus (1.26.0-3) ...
Processing triggers for desktop-file-utils (0.26-1) ...
Processing triggers for mailcap (3.70+nmu1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Errors were encountered while processing:
 mintdrivers
doug@daphne:~/Sandbox$

ichibrosan, 2025-03-10:
I looked around but could not find a way to load the required package.

mtwebster, 2025-03-10:
Please provide more info -

What distro are you using?

The driver manager only works on Linux Mint (not LMDE or anything else). It should also already be installed.

closed 2025-03-16 by ichibrosan